MatterHackers/MatterSlice #5: MatterControl Slice Settings export is not the same as MatterSlice config files

dattas, 2015-01-16:
Hello,

I've written an OctoPrint plugin that will allow users to use MatterSlice with OctoPrint. I've run into an issue where if I export Slice Settings from MatterControl, then try to use that file with -c with MatterSlice, the values don't match up and thus the slice settings are not useful in this instance. Are there plans to either change the settings to be the same values as the slice settings from MatterControl or perhaps a flag to indicate the settings are from MatterControl via a --from-mattercontrol/-mc or something similar?

Should I post this on MatterControl's repo instead?

Thanks!

larsbrubaker, 2015-01-19:
@dattas 

So you are running into some very interesting legacy behavior. We started MatterSlice with Slic3r and our internal settings in the database and therefore the settings files are all slic3r or slic3r styled, but the settings that MatterSlice uses are are translated before being written to the config file. You can find the settings that were sent to MatterSlice in C:\Users\[username]\AppData\Local\MatterControl\data\gcode on windows. You can also find them in the gcode output at the end of the file. You would just need to remove the "; " from the start of each line.

dattas, 2015-01-19:
@larsbrubaker Thanks for the work around on finding that config. Any plans to make this available via a dropdown or something similar? I don't think people will like having to hunt for the config file or parse it out of a sliced file if they want to use MatterSlice outside of MatterControl

larsbrubaker, 2016-05-26:
Yes, we will add this to our road map.

larsbrubaker, 2016-05-26:
Issue moved to [MatterHackers/MatterControl #807](https://github.com/MatterHackers/MatterControl/issues/807) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2016-05-26 by larsbrubaker