MatterHackers/MatterSlice #6: Holes too small

pixel3Design, 2015-02-05:
Hi, and thanks in advance for the gr8 job.
I'm trying to leave Slic3r for MatterSlice, but seems that same issues in maths are present in both.
As in Slic3r, MatterSlice is affected by SMALLER HOLES than STL. As for the Slic3r, external dimensions are correct, only holes are affected. Tryed high and low desity poly holes, different speed, temps, layer height and material density (PP, PLA, ABS, HIPS, PVA, Sugar). Difference is between +0.6 / +0.4 , an unacceptable tolerance. And this is true for Cura, Slic3r, and MatterSlice, but not for SkeinForge, that makes holes correctly.

Don't know big maths, but as far as I can see, there is a constant in holes measures between those softwares, and, if I don't going wrong, is the inset routine, am I right?

I wanna really use your software for production, but I made mechanical parts, and this is a great barrier. Please, don't be stone-deaf like Alexrj. It's not a miscalibration behavior. It's a math fault.

Cheers

larsbrubaker, 2015-02-05:
Thanks for the feedback. This is something we are actually working on. We'll have a fix in a month or so (hopefully). But you can get slightly better tolerances by switching to 'External Perimeters First' under Layres/Perimeters in the latest build.

geneb, 2015-02-21:
I suspect SF might be doing some polyhole mitigation.  See here: http://hydraraptor.blogspot.com/2011/02/polyholes.html

larsbrubaker, 2015-02-21:
That is right. [Here](http://hydraraptor.blogspot.com/2014/06/why-slicers-get-dimensions-wrong.html) is nopheads update that I think is what we will implement. I wish I could fix all these things faster.

nokianhunter, 2016-02-16:
Hi there,
MatterSlice works pretty well for me, too. Thanks for your work.
I wonder if this issue is solved, because I have trouble getting hole dimensions right.
I tried different "Outside Perimeters" settings in the "Filament"->"Extrusion" Tab but it hardly helped.
As this is a problem in many slicers it is probably quite hard to fix.

geneb, 2016-02-16:
On Tue, 16 Feb 2016, nokianhunter wrote:

> Hi there,
> MatterSlice works pretty well for me, too. Thanks for your work.
> I wonder if this issue is solved, because I have trouble getting hole dimensions right.
> I tried different "Outside [...]
The issue isn't a slicer problem.  Check this out:
http://hydraraptor.blogspot.com/2011/02/polyholes.html

If you're having a problem with undersized holes, try small increases 
until you get what you need.

g.

-- 
Proud owner of F-15C 80-0007
http://www.f15sim.com - The only one of its kind.
http://www.diy-cockpits.org/coll - Go Collimated or Go Home.
Some people collect things for a hobby.  Geeks collect hobbies.

ScarletDME - The red hot Data Management Environment
A Multi-Value database for the masses, not the classes.
http://scarlet.deltasoft.com - Get it _today_!

nokianhunter, 2016-02-16:
Thanks for that link, but I already read that, as you posted it before.
This method is just a dirty workaround.
I don't wanna do this manually.
The slicer must be capable of getting the holes correct - at least somehow.
IMHO this is a slicing problem.

larsbrubaker, 2016-02-16:
So we added the ability to set the Outside Perimeters extrusion width. I have been told that this is the same compensation that simplify is doing and it matches well with manual correcting what nophead describes here http://hydraraptor.blogspot.com/2014/06/why-slicers-get-dimensions-wrong.html.

This setting is in our alpha builds now and this new version will be in release this week.

larsbrubaker, 2016-05-26:
This has been released and is in 1.5.

closed 2016-05-26 by larsbrubaker