MatterHackers/MatterSlice #7: No output until 6th layer?

geneb, 2015-02-21:
I'm working on updating the Orion docs and I wanted to show what support and a raft looked like.  Unfortunately, MatterSlice doesn't appear to be doing any extrusion until layer six.
Here's what the raft configuration looks like:
http://www.geneb.org/orion/print-setup.jpg
Here's the layer view showing layer six where the extrusion actually begins:
http://www.geneb.org/orion/layer-six.jpg
This isn't an anomaly in the layer display - the printer actually doesn't begin to extrude until layer six.  Nor does it appear to be doing any moves that would indicate it's laying down a raft.
The build of MatterControl is 1.2.2.3896.

Thanks!

larsbrubaker, 2015-02-21:
Thanks for this bug. I can reproduce it and will fix it asap. Hopefully for our next release (early next week). It looks like the raft just doesn't know what to do when there is no effective surface in the first layer to be printed.

larsbrubaker, 2015-02-21:
So I changed the code to produce a raft even when the first layer of the part has no data. Which is the case with the sharp point of the box. It will never be a very useful raft as there is nothing to really support, but it will let the user have better feedback. This will be in the build next week. Thank you.

closed 2015-02-21 by larsbrubaker

geneb, 2015-02-21:
Thanks a lot Lars.  It dawned on me about 4:30 am that I had no business submitting a bug without more data (gcode, etc).  I'm glad you were able to duplicate it anyway. :)