MatterHackers/MatterSlice #11: Disable retraction for infill?

OhmEye, 2015-04-20:
The Triangle and Hex infill patterns can be very time-inefficient with regard to the high number of short line segments and the resulting high number of retractions. Many printers require retraction settings that incur significant time increases or other performance detriments when the number of retractions per layer is high. For example, bowden extruders often have extruder performance problems performing rapid sequences of closely spaces retractions. On a recent test with a non-bowden extruder where I used retraction distance of 1.5mm at 60mm/s to print a comparison using the same speeds and comparable features to gcode exported from slic3r I found that matterslice gcode took 24% longer to print. The increased layer time was due to hundreds of additional retractions during hex infill.

I suggest providing an option to disable retraction for infill, this would allow for no changes to the pattern itself and significantly improve layer times without much impact to quality since infill is usually not externally visible.

larsbrubaker, 2015-04-20:
We recommend and set the default retraction settings to about 20mm. You do not want to have no retraction on infill because if there is a long move you will get ooze and as you start at the new position you will have no chamber pressure and an inconsistent start to the line. MaterSlice also always retracts when changing islands (moving from one object to another) so small retraction settings are not needed for changing parts.

closed 2015-04-20 by larsbrubaker

OhmEye, 2015-04-20:
I appreciate your response and suggestion, but please consider that sometimes users may have good reason to do things that you do not anticipate. For example, I do not want to increase the minimum distance to 20mm, that would effectively turn off retraction completely on a typical print job. In my specific case, when printing a plate of a hundred small parts filling a 225x320mm build area there are almost no travel moves longer than 20mm because the islands are small. However, the total area of infill on the layer is large and the hundreds of unnecessary retractions while printing infill increases layer time significantly.

Trying to tune minimum distance so that unnecessary infill retractions are minimized is not a practical solution. Not everybody is printing a single large object at a time, and some may also have printers or plastics with an ooze rate for which one minimum distance is acceptable for infill but not for perimeters.

Adding an option to disable retractions for infill would be a simple workaround for the path inefficiency of the triangle and hex patterns due to not being continuous extrusions. Most other slicing engine implementations for infill patterns minimize the number of discontiguous segments so do not have a high number of travels or retractions during infill.