MatterHackers/MatterSlice #22: XY compensation / horizontal size expansion [feature request]

Patola, 2015-09-17:
This is a VERY important feature that deals with fittings on 3d printed parts.

The newer slic3r allows it (XY compensation),
the newer Cura (horizontal size expansion),
and the newer Simplify3D (horizontal size compensation).

Different names but the same concept: amount of mm to expand (positive) or contract (negative) the XY layer.

Its best use is to compensate the border of the most exterior stream of extruded plastic, which is what prevent fittings from working (e.g. a 0.4mm nozzle will have and exterior border of 0.2mm, so you need to compensate by using this setting with -0.2mm).

Please add this feature. The algorithm for the compensation is already there: it's sort of what you use for skirts and brim (the expansion of the exterior lines of the part - only for XY compensation you have to expand interior parts too). Implementing it on the slicing should not be difficult.

gobrien4418, 2015-09-18:
The flip side of this is the expansion of hole diameters; which get 
smaller for the same reason, i.e. the path line runs down the centre of 
a 0.4mm wide extrusion, thus 0.2mm of it is inside the print. Perhaps 
the same or a similar value could allow for expanding holes?

Related and possibly worth considering at the same time is a thermal 
expansion coefficient. Especially with ABS, the whole part shrinks 
something wicked once it comes off the heated bed. It might be nice to 
have an X-Y expansion factor (ratio or percentage) that is automatically 
applied as part of the filament profile?

-- 
oae signature *
Gareth O’Brien*
Director

*Objective Advantage Europe Ltd.*
Dublin, Ireland
+353-76-615-4839 Voice
+353-87-277-9281 Mobile

linkedin <http://ie.linkedin.com/in/garethobrien>
---------------------------------------------------------
This email and the information it contains is confidential and may also 
be privileged. It is intended for the use of the addressee only. If you 
have received this message in error, please notify us immediately by 
replying to this email and delete the message. If you are not the 
intended recipient, you may not use, copy or disclose to anyone the 
message or any information contained in this email. Thank you.

RyushoYosei, 2019-08-06:
I was just looking for this feature as it is a very very important feature.