MatterHackers/MatterSlice #54: Request: stop using an extruder when it's no longer needed

johnlewin, 2016-05-26:
@whosawhatsis commented on [Sun Nov 23 2014](https://github.com/MatterHackers/MatterControl/issues/113)

The slicer should not only be smart enough not to heat an extruder that will not be used (as all others do), but it should also be smart enough to begin cooling down an extruder that will not be needed for the remainder of the print and run the rest of the print as a single-extruder job (disabling dual extruder options like ooze shields and priming towers above that point).

FireflyNJ, 2017-01-15:
**It should be a checkbox to make it optional 'on'.    I want it on.**
Even tho my nozzles are level, it still hits the printed project, when it is hot it does not knock the printed material off the table.
Even if I switch to the other nozzle, then the other nozzle hits, its not a leveling problem, its the fact the the material beads up a little and because nozzle is hot it melts it instead of bumping it.

teejaydub, 2018-01-24:
With a shared nozzle setup, I don't think there's ever a drawback to this, only the benefit of not wasting time and filament on the wipe tower.