MatterHackers/MatterSlice #74: Very slow printing with some parts.

robermeyer, 2016-07-10:
I have a simple part about 2" by 1.5" and 1/4" thick.
Using standard parameters it prints the first layer faster than normal.
Then it print base 3 layers medium speed.
Then when it gets to fill layers it slows to a very slow speed.
When it gets to layer 28 it speeds back up.
It never uses the max speed for anything.
I have printed smaller parts and bigger parts with same settings with no problems.
Whats the problem?

[Stabilizer.zip](https://github.com/MatterHackers/MatterSlice/files/355834/Stabilizer.zip)

unlimitedbacon, 2016-07-11:
There is nothing about that STL that is causing issues. Could you export and post your slice settings? Also, what version are you using?

robermeyer, 2016-07-12:
Sorry to ask but can you tell me how to post my slicer settings.
Do you want me to export my STL to STL?
I am using MatterControl 1.5

larsbrubaker, 2016-07-12:
[Here](http://wiki.mattercontrol.com/Send_Debugging_Information) is a guide to sending debug info. You can add the data to a zip and drop it into your response.

Relevant to this type of bug:
* Model File (you already gave it)
* Slice Settings
* G-Code File (the output that you are seeing as slow)
* Build Number

Thanks in advance!

robermeyer, 2016-07-12:
Hey, thank you guys for looking at this.
I bugs me that some are slow and some are fast.

[Stabilizer.zip](https://github.com/MatterHackers/MatterSlice/files/360388/Stabilizer.zip)

Build 1.5.3.6876

unlimitedbacon, 2016-07-12:
Looks like its slowing down because of the cooling threshold. [This setting](http://wiki.mattercontrol.com/Settings/Filament/Cooling/Cooling_Thresholds/Slow_Down_If_Layer_Print_Time_Is_Below) slows down the printer when necessary in order to give the layers below time to cool. You have it set to 40 seconds. If you bring it down to 10 seconds, it will print at full speed.

Normally this would not be a factor on a print that large, but in your case it is because all of your other speeds are set so high.

larsbrubaker, 2016-07-13:
Assuming this addressed the issue and closing. 

Please let us know if you continue to have problems.

closed 2016-07-13 by larsbrubaker