MatterHackers/MatterSlice #145: Top layer travel Z-Hop

duisenberg, 2016-11-23:
When printing the final top layer, travel moves run in build height, leaving travel marks on the model surface. For some reason - I run retraction with Z Lift - this doesnt seem to apply here.

larsbrubaker, 2016-11-28:
Can you export your profile and attach it to this issue. I will investigate either way but that might help.

larsbrubaker, 2016-11-28:
I am definitely seeing z-hop in the top layer of my test prints. If you can add your settings, the exported gcode and possibly the part as a zip to this issue, that would be a big help. I think the issue your having is probably due to the Minimum Travel Requiring Retraction. Setting this to a much smaller number will likely get you the result you are looking for.

[image]

duisenberg, 2016-11-28:
Might be my settings lead to issues:
[image]

Atm a bit short in time, will try to catch it on image tomorrow.

duisenberg, 2016-11-28:
Found an image that shows the issue, the red marked travel was about 20mm distance. Also there are marks on top but they are very close to 8mm so minimum travel should have matched here.

[image]

duisenberg, 2016-11-30:
Here some more samples:
[image]
[image]
[image]

Print settings same as above. 
Most travels do correctly lead to retraction with Z hop but some are omitted. Didnt see it happen with infill layers, only bottom and top layers. Not a so big issue, but since the Matterslicer really works well on surfaces it would be even better if it wouldnt leave a travel scar.
Using the 1.6.07828 btw. Profile is attached.

[Profile.txt](https://github.com/MatterHackers/MatterSlice/files/622817/Profile.txt)

duisenberg, 2016-12-06:
As for 1.6.1.7967 I didnt see the issue happen any more on top layers.

closed 2016-12-12 by larsbrubaker