MatterHackers/MatterSlice #146: Cannot print both object and support with second extruder

unlimitedbacon, 2016-11-30:
Steps to reproduce:
1. Edit an STL and choose Material 2.
2. Set Support Material Extruder and Support Interface Extruder to 2

In this case the object will still be printed with Material 1, even though Material 2 was chosen. 

[image]

[Related Forum Thread](http://forums.matterhackers.com/topic/378/support-material-extruded-by-ext1-no-matter-what-settings-i-use/3)

jamacof, 2016-11-30:
Actually the object will be printed with material 2, however the support will be printed with material 1 even if its extruder is not heated. Interestingly if you set Support Material Extruder and Support Interface Extruder to 3 it appears to be working fine (looking at the layers display - wasn't able to print due to different problem)

larsbrubaker, 2018-12-29:
This should all be fixed now.

closed 2018-12-29 by larsbrubaker