MatterHackers/MatterSlice #150: Infill overshoot

duisenberg, 2016-12-03:
In some cases infill overshoots the perimeter, even if set to very low overlap:

[image]
[image]
[Knob1.txt](https://github.com/MatterHackers/MatterSlice/files/628931/Knob1.txt)

The txt file contains the STL, Mattercontrol Version 1.6.1.7967

larsbrubaker, 2016-12-03:
Thank you for the report.

Can you export and attach as a zip your profile. It is not required but will be helpful.

duisenberg, 2016-12-06:
[Printer.txt](https://github.com/MatterHackers/MatterSlice/files/633286/Printer.txt)

Sorry, I always search the "export printer profile" menu item. Github atm rejects zips. Well ... txt works.

duisenberg, 2017-09-10:
Since the issue didnt happen again since 1.7 I guess this one can be closed.

closed 2017-09-10 by duisenberg