MatterHackers/MatterSlice #152: If First Layer Height is bigger than Layer Height, MatterSlice will crash

rytz, 2016-12-09:
e.g. Layer Height set to 0.2 and First Layer Height set to 0.4

rytz, 2016-12-09:
Stack trace:

   at System.ThrowHelper.ThrowArgumentOutOfRangeException(System.ExceptionArgument, System.ExceptionResource)
   at System.Collections.Generic.List`1[[MSClipperLib.IntPoint, MatterSlice, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]].get_Item(Int32)
   at MatterHackers.MatterSlice.GrahamScan.GetLowestPoint(System.Collections.Generic.IList`1<MSClipperLib.IntPoint>)
   at MatterHackers.MatterSlice.GrahamScan.GetConvexHull(System.Collections.Generic.List`1<MSClipperLib.IntPoint>)
   at MatterHackers.MatterSlice.PolygonsHelper.CreateConvexHull(System.Collections.Generic.List`1<System.Collections.Generic.List`1<MSClipperLib.IntPoint>>)
   at MatterHackers.MatterSlice.LayerDataStorage.GenerateSkirt(Int32, Int32, Int32, Int32, Int32, Int32, MatterHackers.MatterSlice.ConfigSettings)
   at MatterHackers.MatterSlice.fffProcessor.ProcessSliceData(MatterHackers.MatterSlice.LayerDataStorage)
   at MatterHackers.MatterSlice.fffProcessor.DoProcessing()
   at MatterHackers.MatterSlice.MatterSlice.ProcessArgs(System.String[])
   at MatterHackers.MatterSlice.MatterSlice.Main(System.String[])

larsbrubaker, 2017-01-31:
Fixed.

closed 2017-01-31 by larsbrubaker