MatterHackers/MatterSlice #156: Expand Thin Walls causes blobs in the side of this print

unlimitedbacon, 2016-12-23:
This part has 0.3 mm thick walls. When printed with a 0.5 mm nozzle and Expand Thin Wall turned on, many blobs are created in the sides of the walls.

[Original forum post](https://forums.matterhackers.com/topic/403/poor-slicing-performance-after-1-6-upgrade/3)

[image]

[STL and Printer Profile](https://github.com/MatterHackers/MatterSlice/files/671311/BlobIssue.zip)

larsbrubaker, 2016-12-24:
We have some strange math problems rendering certain features. Dose this also happen while printing?

unlimitedbacon, 2016-12-27:
Not sure. The same artifacts show up in gcode.ws, but it shows them as travel moves.

FireflyNJ, 2017-01-22:
Printing walls thinner then the nozzle size?  I wouldn't expect that to work.

unlimitedbacon, 2017-02-01:
Its definitely not just a rendering error

[image]

larsbrubaker, 2017-02-02:
Will look into improving this.

closed 2017-02-18 by larsbrubaker