MatterHackers/MatterSlice #166: Infill and perimeter speeds not honored if infill speed is 12 or more

tcm0116, 2017-01-18:
Consider a 2 cm test cube with the following speed settings:

   - Infill - 25 mm/s
   - Top Solid Infill - 35 mm/s
   - Inside Perimeters - 40 mm/s
   - Outside Perimeters - 35 mm/s
   - First Layer Speed - 15 mm/s

When slicing with MatterSlice (the version included with version 1.6.2.8214 of MatterControl or the version included with the released 1.6.1 version of MatterControl), the generated perimeter and infill speeds in the G-code for infilled layers do not adhere to the provided settings. This is very apparent when speed is enabled in the layers view of MatterControl and can be seen in the attached image.

If we examine the attached G-code that was generated by MatterSlice, we see that starting at layer 2 (where the non-solid infill starts), the inner wall is printed at 24.4 mm/s, the outer wall is printed at 21.35 mm/s, and the infill is printed at 15.25 mm/s. These are obviously not correct for the settings provided to MatterSlice.

Doing some experimentation, I found that if I set the infill speed to 11 mm/s or less, then the perimeters and infill starting at layer 2 adhere to the prescribed settings.

[Calibration Cube (2cm).gcode.txt](https://github.com/MatterHackers/MatterSlice/files/714618/Calibration.Cube.2cm.gcode.txt)

[image]

unlimitedbacon, 2017-01-18:
I bet this is because of your [minimum layer time](http://wiki.mattercontrol.com/SETTINGS/Filament/Cooling/Cooling_Thresholds/Slow_Down_If_Layer_Print_Time_Is_Below). Try setting it to 0 and see if that changes anything.

tcm0116, 2017-01-26:
@unlimitedbacon you are correct. Changing the minimum layer time to 0 resolved the issue. Looks like I still have a lot to learn!

closed 2017-01-26 by tcm0116