MatterHackers/MatterSlice #231: Bridge identification

doomsed, 2017-06-28:
Matterslice does not identify internal (XY plane) unsupported spans as bridges and are treated as infill (Print speed and fan speed).  Part of the perimeter must be unsupported before it is treated as a bridge .

Unsupported[bridge] perimeters are not treated as bridges, but as perimeters (print speed and fan speed).

Bridging is completely disabled when 'Generate Support Material' is selected: Spans that should either be treated as bridges or have supports generated by 'Support Everywhere' are unsupported and treated as regular perimeters/infill (print speed and fan speed) when 'Generate Support Material' is selected and 'Support Everywhere' is not.

larsbrubaker, 2017-06-29:
Thanks for the through summary.

cwilso03, 2020-05-25:
I think this is a bug, not an enhancement request.  I tried printing a bridging test model from thingiverse (https://www.thingiverse.com/thing:2614805), and the bridge printed at infill speeds.  

After finding this open issue, I recreated the model in Fusion 360 (the attached [Bridging Test.zip](https://github.com/MatterHackers/MatterSlice/files/4677125/Bridging.Test.zip) contains both .f3d and .stl formats), but with open sides, hoping to side-step this problem.  But my model also printed the bridge using infill speeds.

larsbrubaker, 2020-05-31:
Fixed in the next build

closed 2020-05-31 by larsbrubaker

larsbrubaker, 2020-05-31:
[image]

Working much better now. This will ship in the next couple of weeks.