MatterHackers/MatterSlice #295: Request: Cylindrical wipe tower option

teejaydub, 2018-01-27:
For dual extruders to work, SO many things have to be perfect. One of them is that your wipe tower has to not come detached from the build plate and wander into your main object!

I've noticed that when this happens, it's most likely to start at a corner of the wipe tower (just like many objects will first peel up at their corners). So, I think a cylindrical wipe tower would reduce the probability of wipe-tower-related casualties.

teejaydub, 2019-01-25:
Cool, that's great! The link didn't work - maybe it's a private repo? - but I'll look for that. Thanks.

larsbrubaker, 2019-03-23:
This is fixed in code and will be in the next build (should be mid April)

closed 2019-04-02 by larsbrubaker