MatterHackers/MatterSlice #398: Fractured Gcode file please help

martinpirringer, 2018-10-18:
On 3 occasions now I had a print "fail" due to incomplete gcpde file. I need to use the gcode as the computer connected to the pulse cannot slice that many layers (still the layer problem) I got to slice it on the computer with 16 gb of ram etc. Unfortunately it seems that when you hit export it does not always copy the whole gcode file and then somewhere in the print it just stops cause it ran out of gcode. I cannot load the Gcode file on the older computer as you cannot properly print a gcode file as it seems to mess with the Z offset (mine is -.4 so if I run a previously saved gcode file I engrave the first layer on the build plate. So I cant look at the screen in MC if everything is there and got to use the Vicky. It prints the offset correctly there hence my guess that MC reapplies the offset and/or bed leveling. (is there a workaround?) For now I can only suggest that once you write the gcode to the microsd card you open it in a text editor and go to the bottom and check for the "Mattercontrol completed successfully" message and the commands that turn everything off and move the build plate forward etc I asked support if they know a workaround but the suggesgted to raise it here so hence I did. It is quite complicated at the moment to make tall prints this one is about 165mm tall. I wish you guys fix the layer limit soon so it slices all layers on all machines. If there is a machine that shouldnt be used then it should be in the MC specs and MC should say something like "this machine has not enough ram/processors whatever". All I want to do is get those prints done I need to get done. So until you fix it some work around or alternate procedures would be apreciated.

jlewin, 2018-10-18:
Can you attached the partial gcode file for review?

martinpirringer, 2018-10-22:
Update: It seems to never happen if you just got into MC and Load stl - then hit export and give the file a name that does not exist yet in that directory. But seems to happen at times if you slice first then hit export and save it to a file that already exists. Example I save a file as mypart.gcode then maybe the print requires some changes so I create a new stl hit "clear bed" load stl, then slice, look if everything is where it is supposed to be and then hit export and save it again to the existing "mypart.gcode" file by clicking on it when the directory displays. It does not happen everytime but about 1 in 5 over the past couple of days. It might also be that I might have clicked something in MC to go to the next project. It would be great maybe to refuse keyboard/mouse input if its critical except maybe a cancel and pop up a message that says "export completed successfully" when its done and the file has been written to and closed and flushed.

martinpirringer, 2018-10-22:
Also hope you got the gcode files