MatterHackers/MatterSlice #399: Mattercontro; can be started multiple times

martinpirringer, 2018-10-18:
MatterControl should only be able to be started once. You can use a mutex to achieve that through windows. If minimized a user might start it again from the icon on the desktop or somewhere and now you got 2 mattercontrols potentially sending commands to the printer

unlimitedbacon, 2018-10-18:
This is intentional. Opening multiple instances of MatterControl allows you to run multiple printers from the same computer. Windows blocks multiple programs from using the serial port at once, so there is no risk of two instances sending commands to the same printer.

In the future, the plan is to allow a single instance of MatterControl to control multiple printers. Each printer will have it's own tab. However, there is still some major work that needs to be done in order to make this possible.

Also, this is a MatterControl issue so it should be reported in the MatterControl repository, not MatterSlice.

closed 2018-10-18 by unlimitedbacon

martinpirringer, 2018-10-20:
Update: It seems to never happen if you just got into MC and Load stl - then hit export and give the file a name that does not exist yet in that directory. But seems to happen at times if you slice first then hit export and save it to a file that already exists. Example I save a file as mypart.gcode then maybe the print requires some changes so I create a new stl hit "clear bed" load stl, then slice, look if everything is where it is supposed to be and then hit export and save it again to the existing "mypart.gcode" file by clicking on it when the directory displays. It does not happen everytime but about 1 in 5 over the past couple of days. It might also be that I might have clicked something in MC to go to the next project. It would be great maybe to refuse keyboard/mouse input if its critical except maybe a cancel and pop up a message that says "export completed successfully" when its done and the file has been written to and closed and flushed.

unlimitedbacon, 2018-10-22:
@martinpirringer, I don't know what you are talking about. I think you have commented on the wrong bug.

martinpirringer, 2018-10-22:
Sorry must have clicked on the wrong email link am new to Github and can't find my bugs except by clicking on the email link This goes with the one where it corrupts the Gcode file. Now If I can find it I copy and paste it there