MatterHackers/MatterSlice #617: MatterSlice DumpSettings outputs settings that cannot be read in later

scherej958, 2022-05-24:
The DumpSettings() function in ConfigSettings.cs outputs settings that cannot be read back in using ReadSettings(). This is caused by two issues: (1) Setting only has a get accessor (i.e. CoastAtEndDistance_um) and (2) Comments/Descriptions are appended to the setting value.

I have corrected:
(1) adding a new private function IsSettable() and calling between setting "object value" and "switch(property.PropertyType.Name) in DumpSettings() function.

private bool IsSettable(PropertyInfo property)
{
	string name = property.Name;
	MethodInfo[] mi = property.GetAccessors();
	foreach (MethodInfo info in mi)
	{
		if (info.Name.Contains("set_"))
					return true;
	}
	return false;
}

public void DumpSettings(string fileName)
{
	...
	string name = property.Name;
	object value = property.GetValue(this);

	// JGS 5/23/22 Changed to remove settings that cannot be read in later
	if (IsSettable(property) == false)
		continue;
	///////////////////////////////////////////////////////////////////////

	switch (property.PropertyType.Name)
	{
	...
}

(2) Add the following lines to SetSetting() function:
public bool SetSetting(string keyToSet, string valueToSetTo)
{
	...
	// Drop quotes from all other settings
	valueToSetTo = valueToSetTo.Replace("\"", "").Trim();
	// JGS 5/23/22 Added to strip comments from values
	// Drop all comments for the setting value
	if (valueToSetTo.Contains("# ") == true)
		valueToSetTo = valueToSetTo.Substring(0, valueToSetTo.IndexOf('#'));
	///////////////////////////////////////////////////////////////////////
	...
}

These changes will allow input of the DumpSettings() output, as a valid configuration file. It effectively removes all "xxx_um" settings from the DumpSettings output and allows for comment fields beginning with "# " to be removed from the setting value. It appears that the "xxx_um" settings are all derived from other settings and therefore should not be available for input.

scherej958, 2022-05-24:
I am not complaining about existing functionality. Just wanted to pass on a potential fix. I have added it to my code but thought I would pass it back for consideration.

larsbrubaker, 2022-07-01:
Code added for next release. Thank you.

closed 2022-07-01 by larsbrubaker