MatterHackers/MatterSlice #618: MatterSlice settings "RaftSurfaceLayers" and "RaftSurfacePrintSpeed" are not able to be read into configuration

scherej958, 2022-05-24:
The "RaftSurfaceLayers" and "RaftSurfacePrintSpeed" will generate errors when attempting to read in from configuration file. The issue is that both settings have get accessors only. I have corrected this issue with the following code:
[ConfigSettings.cs]

"ReadSurfaceLayers":
CHANGED LINE - public int RaftSurfaceLayers => 2;   to   public int RaftSurfaceLayers { get; set; }
[ public void SetToDefault() ]
ADDED LINE - RaftSurfaceLayers = 2;

"RaftSurfacePrintSpeed":
CHANGED LINE - public int RaftSurfacePrintSpeed => RaftPrintSpeed;   to   public int RaftSurfacePrintSpeed { get; set; }
[ public void SetToDefault() ]
ADDED LINE - RaftSurfacePrintSpeed = RaftPrintSpeed;

I believe that both of these functions should be supported to be used within the configuration file.

scherej958, 2022-05-24:
I am not complaining about existing functionality. Just wanted to pass on a potential fix. I have added it to my code but thought I would pass it back for consideration.

larsbrubaker, 2022-07-01:
Fixed in next release

commit 58f13d68 (2016-05-07): Fixed #618 the bug where air gap was not applied when there were no interface layers.
Will be available in release 1.6