MatterHackers/MatterSlice #619: MatterSlice generates an Exception if ModelMatrix is read in the configuration file

scherej958, 2022-05-24:
The ModelMatrix setting does not appear to be supported in the configuration file read function "ReadSettings(). The issue appears to be that "SetSettings()" function does not support PropertyType = "Matrix4X4". To address this issue I have added the following code:

[ConfigSettings.cs]

		// JGS 5/23/22 - Added to satisy the ModelMatrix setting value
		case "Matrix4X4":
			{
				string[] setVars = valueToSetTo.Split(',');
				double[] setVals = new double[setVars.Length];
				for(int i = 0; i < setVars.Length; i++)
                                        Double.TryParse(setVars[i], out setVals[i]); 
				property.SetValue(this, new Matrix4X4(setVals));
			}
			break;

This code has been added between the "case FMatrix3x3:" and "case "DoublePoint:" case statements in the "SetSetting()" function.
NOTE: The input line in the configuration file might need re-formatting from the settings.ini output to place all values of matrix on one line:  "ModelMatrix=1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1". The current settings.ini file has the matrix values on separate lines.

scherej958, 2022-05-24:
I am not complaining about existing functionality. Just wanted to pass on a potential fix. I have added it to my code but thought I would pass it back for consideration.

larsbrubaker, 2022-06-09:
We'll get this fixed. Thanks.

larsbrubaker, 2022-07-01:
Added for next release. Thank you.

closed 2022-07-01 by larsbrubaker